robopeak/rplidar_arduino #5: error：No such file or directory？

lishuzhou, 2016-04-24:
I met the error when i compile the code ，what should i do?
[image]
[image]

lishuzhou, 2016-04-24:
[image]

jabrena, 2016-04-24:
You should bind the library to the project.

Read this post:
https://www.arduino.cc/en/Guide/Libraries

lishuzhou, 2016-04-26:
thanks a lost! I have soved it .I found that it is beause of the version of the arduino IDE.I use the old version ,after that it is no error!

lishuzhou, 2016-05-21:
I download this code into the arduino  due ,but the arduino due did not run,what should Do?

AliEzzat383, 2023-06-28:
compress RPLidarDriver file (right click > add to archive > select zip format) then in Arduino IDE (sketch>include Library > add.zip library) then select RPLidarDriver.zip file that you created