robopeak/rplidar_arduino #13: improper data with Arduino

Jerwinprabu, 2019-03-03:
I have tried the simple_connect example with Arduino Mega. Because I can't use Arduino UNO for checking results. Here, I have tried to use Serial1 for checking lidar data, but the data is improper. 

Output:

 theta: 200.25 Dist: 01659.00 Q: 47 
   theta: 50.80 Dist: 01700.00 Q: 47 
   theta: 340.80 Dist: 02213.00 Q: 47 
   theta: 73.33 Dist: 02208.00 Q: 47 
   theta: 211.89 Dist: 02222.00 Q: 47 
   theta: 111.44 Dist: 02229.00 Q: 47 
   theta: 79.00 Dist: 02234.00 Q: 47 
   theta: 204.55 Dist: 02238.00 Q: 47 

Is there any way to solve this problem?

Titibo26, 2019-06-26:
i can't see what is improper ? I got the same.

Jerwinprabu, 2019-06-26:
Can you check the output carefully. The data is not continuous data. 

If the first data is theta: 200.25 Dist: 01659.00 Q: 47 then next it should be 201, 202, 203 etc. But it is directly from 200 degree to 50 degree.

It is a random data. It should give the data from 0 degree to 360 degree. So I am asking solution for getting the proper data.

Titibo26, 2019-06-28:
Yes i know, i have the same output. I'm not sur if it's a problem. it seems to be the normal behaviors because in the window app "fram_grabber" wich is given in the source code, they sort the array of point by the angle value. It is not clear how the lidar should work and i can't find anthing about this, but a solution is to sort output value by yourself in your code (as the arduino library doesn't do it).

It would be way better to have output allready sorted by angle... Maybe its a problem related to the serial link, if the µC can't handle so much and so fast data some are deleted in the buffer ? 

You can try to contact slamtech here : support@slamtec.com they are fast but its really hard to have real answer on their products...

If you find a solution, post it here ! :)

Pilooz, 2019-09-17:
Which kind of Lidar do you have ?
A1 or A2 ?

Jerwinprabu, 2019-09-18:
RP LIDAR A1M8 360 Degrees Laser Range Finder- 6 Meter Range

gunnicus, 2022-01-11:
> RP LIDAR A1M8 360 Degrees Laser Range Finder- 6 Meter Range

Hello sir, do you mind share some details how to connect arduino mega to RP LIDAR? I'm using PR LIDAR A1M6 REV6, I can't get any data from the lidar to arduino, all I got is timeout error.

Jerwinprabu, 2022-01-11:
@gunnicus 
[image]

gunnicus, 2022-01-11:
> @gunnicus Suggesting you to use Raspberry pi 4 (2/4/8 GB RAM) [image]

Thanks you so much sir for your reply..it looks like my rplidar is a bit different compare to yours
I followed this diagram but never got it working...I'm not sure if this diagram is correct. This is from their A1M8 datasheet...
[image]

Jerwinprabu, 2022-01-11:
@gunnicus The New RP Lidar is a compact version but the configurations and protocol is same as the old Version. You can try this

[image]

gunnicus, 2022-01-11:
> @gunnicus The New RP Lidar is a compact version but the configurations and protocol is same as the old Version. You can try this
> 
> [image]
> 
> https://github.com/robopeak/rplidar_arduino

Thank you sir, that's very kind of you. Yes basically I did that, but no luck, I can spin the lidar if I write  analogWrite(RPLIDAR_MOTOR, 255) inside loop(), but all the commands timed out...

Jerwinprabu, 2022-01-11:
@gunnicus Can you specify your Application or you can share the code to check the issue, It will help me to provide the proper suggestion.

gunnicus, 2022-01-11:
> @gunnicus Can you specify your Application or you can share the code to check the issue, It will help me to provide the proper suggestion.

Hello Sir, please see the attached pitcure it is my wireing. The code is simple_connect example, the only change I made is to use Serial1 instead of Serial : lidar.begin(Serial1); . Thank you so much for your help.

[image]

gunnicus, 2022-01-12:
> @gunnicus Can you specify your Application or you can share the code to check the issue, It will help me to provide the proper suggestion.

Hello sir, I did more test, the issue seems to be the power supply for the core. I measured the voltage between the vcc5v and groud, the voltage is only 4.1v, I checked the spec it says min voltage is 4.9v. If you run the lidar with Arduino mega, what is the voltage supplied to the core interface?
[image]
[image]

Jerwinprabu, 2022-01-12:
@gunnicus You can use external power supply for 5v and make a Common Ground between 2 different source.

Pilooz, 2022-01-12:
Hi,
I confirm what @gunnicus said. Arduino does not deliver enought power to lead the lidar.
Consider using an external poxer supply in the good way. For example you could use a 5/5A (which is 25W) block.

gunnicus, 2022-01-28:
> 200.25 Dist: 01659.00 Q: 4

Hello sir, I finially got the lidar working with Arduino using external power as you suggested. 
Now I'm facing the problem you had, the data I received are not continues, and I received lots data with Quality  = 0, does it mean I should discard the data if the Quaality is too low?
Distance : 4160.00, Angle : 162.09, Quality : 0
There are also data the Angle is greater then 360 degree, does it mean those data are also bad data or I need to convert the Angle make it within the range between 0 - 360?
Distance :3969.00, Angle : 492.42,Quality : 32
Thank you very much.

gunnicus, 2022-01-28:
> @gunnicus You can use external power supply for 5v and make a Common Ground between 2 different source.

Hello sir, I finially got the lidar working with Arduino using external power as you suggested.
Now I'm facing the problem you had, the data I received are not continues, and I received lots data with Quality = 0, does it mean I should discard the data if the Quaality is too low?
Distance : 4160.00, Angle : 162.09, Quality : 0
There are also data the Angle is greater then 360 degree, does it mean those data are also bad data or I need to convert the Angle make it within the range between 0 - 360?
Distance :3969.00, Angle : 492.42,Quality : 32
Thank you very much.

Titibo26, 2022-01-28:
If angle = 492
Juste do angle = angle - 360.

About the qualiy you need to check in the datasheet, its Witten. 

The best thing you Can Do is to draw scan points and Sees if it match reallity

siemeT, 2023-01-17:
> @gunnicus The New RP Lidar is a compact version but the configurations and protocol is same as the old Version. You can try this
> 
> [image]
> 
> https://github.com/robopeak/rplidar_arduino

I have my setup connected like this, but the data I'm getting seems useless. I shouldn't be getting so little measurements between startbits, right? and I guess this 'quality' is also too low. and if I plot the data there is always somehow a circle, a horizontal line and a diagonal line in it that shouldn't be there.  

does anybody have a clue about what could be the issue? (I also tried with an lab power supply connected to VCC_5V, but that gave exactly the same results. The hardware seems fine. Because if I connect using the usb connector and framegrabber then the output looks ok. 

[degree;distance;quality;startbit]
124.20;8531.25;9;0
270.11;1814.25;15;0
124.80;8866.25;15;0
278.28;1763.00;15;0
281.84;3813.50;15;0
289.00;1275.25;15;0
290.23;1256.00;15;0
37.88;14415.50;36;0
125.00;9516.25;15;0
207.16;12517.50;18;1
0.05;1103.50;15;0
306.67;9893.75;15;0
310.30;3971.50;15;0
318.47;919.25;15;0
319.72;3971.50;15;0
124.09;3971.25;61;1
327.28;3971.25;6;1
27.34;7503.50;41;0
27.30;15311.50;42;1
55.34;463.50;17;1
77.36;3907.50;37;1
358.11;943.50;15;0
359.25;13047.00;15;0
31.25;2383.50;15;0
17.44;384.00;15;0
20.94;0.00;0;0
22.11;0.00;0;0
1.28;128.00;0;0
5.61;894.25;0;0
23.00;0.00;11;0

MarwanHashemi00, 2023-04-01:
> Hello sir, I finially got the lidar working with Arduino using external power as you suggested. 

@gunnicus hello bro, I am currently stuck with this problem. My rplidar is not spinning when I upload the code, and I even supplied my arduino with 12 volts external and still no spinning. How did you manage to solve the problem? Could you post a pic of your setup if you still have it?

gunnicus, 2023-04-01:
> Hi, Arduino cannot supply enough voltage to the lidar, I think what I did was connect the lidar vcc and ground to a USB, and connect the ground to arduino ground.

waseemR02, 2023-04-18:
@siemeT I am working with the A1M8 model and I m stuck the same with data being really inconsistent. Are we supposed to sort the angles later on, between each startBit. If so, then even startBits are not consistent, as in they appear very frequently, not in intervals of 360. 
If you had any break-though with this, can you let me know?

gimt29370120, 2025-08-14:
I originally processed every scan point and logged them to the SD card.
Then I tried processing every 3rd point instead, filtering out points with quality = 0 or distance = 0, and everything worked perfectly.
`static int counter = 0;
if (IS_OK(lidar.waitPoint())) {
    counter++;
    if (counter % 3 == 0) {
        // perform data processing here...
    }
}`

[image]

[image]