robopeak/rplidar_arduino #17: RPLIDAR A1 with Arduino Leonardo

YouMinJung, 2021-01-05:
Hi.
I want to connect RPLIDAR with arduino leonardo.
So I tried to use below code.

```
#include <RPLidar.h>

// You need to create an driver instance
RPLidar lidar;

#define RPLIDAR_MOTOR 3 // The PWM pin for control the speed of RPLIDAR's motor.
                        // This pin should connected with the RPLIDAR's MOTOCTRL signal
bool transmit = false;
byte incomingByte;

void setup() {
  // bind the RPLIDAR driver to the arduino hardware serial
  Serial.begin(9600);
  lidar.begin(Serial1);
 
  Serial.begin(115200);
  while (!Serial) {
    ; //
  }

  // set pin modes
  pinMode(RPLIDAR_MOTOR, OUTPUT);
}

void loop() {

  if (IS_OK(lidar.waitPoint())) {
    float distance = lidar.getCurrentPoint().distance; //distance value in mm unit
    float angle    = lidar.getCurrentPoint().angle; //anglue value in degree
    bool  startBit = lidar.getCurrentPoint().startBit; //whether this point is belong to a new scan
    byte  quality  = lidar.getCurrentPoint().quality; //quality of the current measurement
    if (Serial.available() > 0) {
      incomingByte = Serial.read();
      transmit = !transmit;
    }
    Serial.println(distance);
    //perform data processing here...

  } else {

    analogWrite(RPLIDAR_MOTOR, 0); //stop the rplidar motor

    // try to detect RPLIDAR...
    rplidar_response_device_info_t info;

    if (IS_OK(lidar.getDeviceInfo(info, 100))) {

      Serial.println("Detected");
      // detected...
      lidar.startScan();
      analogWrite(RPLIDAR_MOTOR, 255);
      delay(1000);
    }
  }
}
```

When I executed this program, RPLIDAR didn't work.
The code is stuck in the "else" part.
When I did "analogWrite(RPLIDAR_MOTOR, 255);" motor rotated well but data was empty.
Maybe RPLIDAR didn't connected... "if (IS_OK(lidar.getDeviceInfo(info, 100)))" this part is "False".
How can I fix this problem??

rsilverioo, 2021-06-14:
@YouMinJung Did you manage to solve this? I'm stuck with the same problem 😅

gunnicus, 2022-01-11:
> Hi. I want to connect RPLIDAR with arduino leonardo. So I tried to use below code.
>
> ```
> #include <RPLidar.h>
>
> // You need to create an driver instance
> RPLidar lidar;
>
> #define RPLIDAR_MOTOR 3 // The [...]

Hello sir, did you find a solution for this? I'm having same problem.

gunnicus, 2022-01-11:
> Hello sir, did you find a solution for this? I'm having same problem.

Hello sir, did you find a solution for this? I'm having same problem.